wuphilipp/gello_software #1: Timeout in Robot Server

AlchemicRonin, 2023-11-25:
After running `python experiments/launch_nodes.py --robot=xarm` in the terminal, the program fails to control xArm and its output is shown below.
[image]
Also, running `python experiments/run_env.py --agent=gello` changes nothing.

AlchemicRonin, 2023-11-25:
Problem solved.

closed 2023-11-25 by AlchemicRonin

ruanjinchen, 2024-01-28:
Hello, I'm sorry to bother you. I would like to ask you where this problem occurred and how you solved it. I also encountered it.

dayong233, 2025-04-22:
> 问题解决了。

Hello, I also encountered this problem. May I ask how you solved it?

Navaneethakrishnan-S-M, 2025-07-23:
Hey @dayong233 and @ruanjinchen — hope you both managed to get it working by now, since it's been a while. But just in case it's still useful for anyone:

I ran into the same timeout issue:

```
Timeout in Robot Server, Robot: <gello.robots.xarm_robot.XArmRobot object at 0x7f596fd21720>
```

What worked for me was simply running:

```bash
python experiments/run_env.py --agent=gello
```

This resolved the timeout and got things moving. Hope it helps!