wuphilipp/gello_software #15: "Timeout in ZMQLeaderServer serve" when trying to test gello panda in MuJoCo

R8ndeer, 2024-06-26:
Hi, I was trying to configure gello for the Franka Panda and reached the following step:

"
First test your GELLO with a simulated robot to make sure that the joint angles match as expected. In one terminal run
`python experiments/launch_nodes.py --robot <sim_ur, sim_panda, or sim_xarm>`
"
Then I got this error:
`Timeout in ZMQLeaderServer serve`

It seems that the server is waiting for some "stop action" to be set? I am very confused about this. If anyone has an idea why this happens or has done simulation of panda in mojoco, could you please share your thoughts on this?

Thank you very much!

R8ndeer, 2024-06-27:
It seems that `self._socket.recv()` always times out

zj30, 2024-07-07:
Hi,I have the same confusion as you, have you solved it?

R8ndeer, 2024-07-07:
> Hi,I have the same confusion as you, have you solved it?

Yes! It turns out I just needed to run the controller as a separate process in a new terminal.

After launching the simulation, in a new terminal run the controller nodes:
`python experiments/run_env.py --agent=gello`
and it should work! I could control the arm in mujoco using gello.

zj30, 2024-07-15:
> Yes! It turns out I just needed to run the controller as a separate process in a new terminal.
>
> After launching the simulation, in a new terminal run the controller nodes: `python [...]

Thank you for your help,I finally find that the initial angle of the arm was set incorrectly,after fix it, I could control the arm by following your advice!

closed 2024-08-01 by R8ndeer