wuphilipp/gello_software #18: Stuck at attempting to connect

osamahesham357, 2024-07-31:
The code was working fine, then it started to stuck at attempting to connect, whatever I do it stuck  there.
[image]

wuphilipp, 2024-08-07:
Humm unfortunatly I have not encountered this issue, so don't have any concreate advice. Do you have more detailed diagnostics? Have you tried restarting the system or the dynamixels?

OH357, 2024-08-07:
Yes i tried restarting, but sometimes it works, i think the issue is to get
the connection up to the simulation or the real robot, if the simulation or
real robot has an error and didn't run correctly, this happens.
I also have another question any way to run both simulation and real
robotat same time?
OH

fireman5379, 2025-04-28:
Same! have you solved it? @OH357

OH357, 2025-04-28:
It was an IP issue while connecting to the real robot, I found it weird as it kept mentioning it's a usb serial port issue but after debugging it was an issue with the IP.