wuphilipp/gello_software #40: ROS2 for UR

yoann-fleytoux, 2025-06-20:
Hi all, thanks for your great work. I m building a gello for my UR10e, I saw the fr3 ros2 package, I was wondering if there were any plans to provide similar things for the UR ?

JemuelStanley47, 2025-06-28:
🙋‍♂️ Not one of the authors, but I’ve been actively using GELLO with my UR5e and UR10e via ur_rtde. While it would be great if the maintainers bring in official support for Universal Robots (like what's available for Franka), I’m currently working on integrating GELLO with the [ROS 2 UR driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver) for my own use.

JemuelStanley47, 2025-06-30:
Hi @yoann-fleytoux, quick update. I got a **minimalistic implementation working** 🎉

🔗 **Repo**: [jemuelstanley47/gello_software_UR (branch: `jemprem/ur`)](https://github.com/JemuelStanley47/gello_software_UR/tree/jemprem/ur)  
📄 [CHANGELOG.md](https://github.com/JemuelStanley47/gello_software_UR/blob/062f96e46dc89673cbf7a42f7a4e69e284b64f42/CHANGELOG.md) has details on the key changes

### Highlights:
- ✅ Separate ROS 2 (humble) package: `ur_gello_state_publisher`
- 🔁 Uses `scaled_joint_trajectory_controller` from UR driver
- 🔄 Switched to **Gello Agent** interface (vs Dynamixel in Franka support)
- 🎯 Verified with **URSim** and **IsaacSim**  

🎥 **URSim Demo**  
https://github.com/user-attachments/assets/ca3d0913-f90b-480c-894a-f0d92f2daa6e

🎥 **IsaacSim Demo**  
https://github.com/user-attachments/assets/4299cce4-1d48-47d8-9ad8-7e12d880955a
Note:   *(IsaacSim video is jerky due to untuned assets)*
> ⚠️ Gripper integration not yet implemented — I plan to follow a structure similar to Franka (URDF + launch + publisher). Will update soon.

Appreciate any inputs!

yoann-fleytoux, 2025-07-01:
I will try it on my ur10e  Wednesday/Thursday :)

foreignlands, 2025-08-28:
> 我将在周三/周四在我的 ur10e 上尝试一下:)

你好，请问你这边尝试成功了么：
我这边/gello/joint_states 有话题发布，但是
 /joint_trajectory_controller/joint_trajectory 话题始终为空

激活轨迹控制器也已经激活了

foreignlands, 2025-08-28:
> 你好[@yoann-fleytoux](https://github.com/yoann-fleytoux)，快速更新。我得到了一个**极简的实现**🎉
> 
> 🔗 **Repo** : [jemuelstanley47/gello_software_UR (分支: `jemprem/ur`)](https://github.com/JemuelStanley47/gello_software_UR/tree/jemprem/ur) 📄 [CHANGELOG.md](https://github.com/JemuelStanley47/gello_software_UR/blob/062f96e46dc89673cbf7a42f7a4e69e284b64f42/CHANGELOG.md)包含主要变更的详细信息
> 
> ### 亮点：
> * ✅ 单独的 ROS 2（humble）软件包：`ur_gello_state_publisher`
> * 🔁`scaled_joint_trajectory_controller`来自 UR 驱动程序的用途
> * 🔄 切换到**Gello Agent**界面（与 Franka 支持中的 Dynamixel 相比）
> * 🎯 已通过**URSim**和**IsaacSim验证**
> 
> 🎥 **URSim 演示**
> 
>  gello_ursim.mp4 
> 🎥 **IsaacSim 演示**
> 
>  gello_isaacsim.mp4 
> 注意：（ _由于未调整资产，IsaacSim 视频很不流畅）_
> 
> > ⚠️夹具集成尚未实现——我计划采用与作者 Franka 软件包类似的结构。很快就会更新。
> 
> 感谢任何意见！

你好，我想咨询一些问题，
首先我修改了一下你的代码，以适应我的端口号，
然后我大致使用了这些命令，但是仿真机械臂始终没有动作：

docker run --rm -it -p 5900:5900 -p 6080:6080 --name ursim universalrobots/ursim_e-series
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=172.17.0.2
ros2 control switch_controllers --activate joint_trajectory_controller --deactivate forward_position_controller forward_velocity_controller

sudo chmod 666 /dev/ttyUSB0
export PYTHONPATH=$PYTHONPATH:/home/summer/gello_software_UR/third_party/DynamixelSDK/python/src

ros2 launch ur_gello_state_publisher main.launch.py com_port:=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0
ros2 launch ur_gello_state_publisher send.trajectory.launch.py 

我这边/gello/joint_states 有话题发布，但是
/joint_trajectory_controller/joint_trajectory 话题始终为空

激活轨迹控制器也已经激活了

foreignlands, 2025-08-28:
> I will try it on my ur10e Wednesday/Thursday :)

@yoann-fleytoux 
Hello, have you successfully tried it on your end
I have a topic published in /gello/joint_states, but
The topic of '/joint_trajectory_controller/joint_trajectory' remains empty
The activation trajectory controller has also been activated

foreignlands, 2025-08-28:
> 你好[@yoann-fleytoux](https://github.com/yoann-fleytoux)，快速更新。我得到了一个**极简的实现**🎉
>
> 🔗 **Repo** : [jemuelstanley47/gello_software_UR (分支: [...]
@JemuelStanley47 

Hello, I would like to inquire about some issues,
First, I modified your code to adapt to my port number,
Then I roughly used these commands, but the simulated robotic arm never moved:
```
docker run --rm -it -p 5900:5900 -p 6080:6080 --name ursim universalrobots/ursim_e-series
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=172.17.0.2
ros2 control switch_controllers --activate joint_trajectory_controller --deactivate forward_position_controller forward_velocity_controller
sudo chmod 666 /dev/ttyUSB0
export PYTHONPATH=$PYTHONPATH:/home/summer/gello_software_UR/third_party/DynamixelSDK/python/src
ros2 launch ur_gello_state_publisher main.launch.py com_port:=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0
ros2 launch ur_gello_state_publisher send.trajectory.launch.py
```
I have a topic published on 
/gello/joint_states, 
but the topic on 
/joint_trajectory_controller/joint_trajectory 
remains empty
The activation trajectory controller has also been activated