wuphilipp/gello_software #49: Frank ros2 can't get  /franka_robot_state_broadcaster/current_pose infomation

Jinming-Li, 2025-07-23:
while I run : ros2 launch franka_fr3_arm_controllers franka_fr3_arm_controllers.launch.py robot_ip:=192.168.1.13 load_gripper:=false

>[INFO] [launch]: All log files can be found below /home/midea/.ros/log/2025-07-23-14-51-34-525380-midea-30813
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [joint_state_publisher-1]: process started with pid [30814]
[INFO] [robot_state_publisher-2]: process started with pid [30816]
[INFO] [ros2_control_node-3]: process started with pid [30818]
[INFO] [spawner-4]: process started with pid [30820]
[INFO] [spawner-5]: process started with pid [30822]
[INFO] [spawner-6]: process started with pid [30824]
[ros2_control_node-3] [WARN] [1753253494.658950720] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] text not specified in the prefix tag
[ros2_control_node-3] text not specified in the arm_prefix tag
[ros2_control_node-3] [INFO] [1753253494.659122168] [resource_manager]: Loading hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1753253494.664227462] [resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1753253494.664275957] [FrankaHardwareInterface]: Connecting to robot at "192.168.1.13" ...
[ros2_control_node-3] [INFO] [1753253494.671875916] [FrankaHardwareInterface]: Successfully connected to robot
[ros2_control_node-3] [INFO] [1753253494.675433669] [service_server]: Service started
[robot_state_publisher-2] [INFO] [1753253494.684187982] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1753253494.684306525] [robot_state_publisher]: got segment fr3_link0
[robot_state_publisher-2] [INFO] [1753253494.684317583] [robot_state_publisher]: got segment fr3_link1
[robot_state_publisher-2] [INFO] [1753253494.684324849] [robot_state_publisher]: got segment fr3_link2
[robot_state_publisher-2] [INFO] [1753253494.684331325] [robot_state_publisher]: got segment fr3_link3
[robot_state_publisher-2] [INFO] [1753253494.684337648] [robot_state_publisher]: got segment fr3_link4
[robot_state_publisher-2] [INFO] [1753253494.684344005] [robot_state_publisher]: got segment fr3_link5
[robot_state_publisher-2] [INFO] [1753253494.684350074] [robot_state_publisher]: got segment fr3_link6
[robot_state_publisher-2] [INFO] [1753253494.684356406] [robot_state_publisher]: got segment fr3_link7
[robot_state_publisher-2] [INFO] [1753253494.684362470] [robot_state_publisher]: got segment fr3_link8
[joint_state_publisher-1] [INFO] [1753253494.764318048] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[spawner-5] [INFO] [1753253494.772257727] [spawner_franka_robot_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-6] [INFO] [1753253494.772258113] [spawner_joint_impedance_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-4] [INFO] [1753253494.772401682] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-3] [INFO] [1753253494.783232015] [action_server]: Action server started
[ros2_control_node-3] [INFO] [1753253494.783277217] [resource_manager]: Successful initialization of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1753253494.783409185] [resource_manager]: 'configure' hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1753253494.783419408] [resource_manager]: Successful 'configure' of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1753253494.783434918] [resource_manager]: 'activate' hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1753253494.783602557] [FrankaHardwareInterface]: Started
[ros2_control_node-3] [INFO] [1753253494.783609730] [resource_manager]: Successful 'activate' of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1753253494.842339339] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-3] [INFO] [1753253494.842379604] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-3] [WARN] [1753253494.846552282] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-3] [INFO] [1753253495.024029886] [controller_manager]: Loading controller 'joint_impedance_controller'
[ros2_control_node-3] [INFO] [1753253495.028435948] [controller_manager]: Loading controller 'franka_robot_state_broadcaster'
[spawner-6] [INFO] [1753253495.028884672] [spawner_joint_impedance_controller]: Loaded joint_impedance_controller
[ros2_control_node-3] [INFO] [1753253495.032351975] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-5] [INFO] [1753253495.032717800] [spawner_franka_robot_state_broadcaster]: Loaded franka_robot_state_broadcaster
[ros2_control_node-3] [INFO] [1753253495.036304906] [controller_manager]: Configuring controller 'joint_impedance_controller'
[spawner-4] [INFO] [1753253495.036858623] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1753253495.040541613] [controller_manager]: Configuring controller 'franka_robot_state_broadcaster'
[ros2_control_node-3] [INFO] [1753253495.065433144] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-3] [INFO] [1753253495.065519867] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ros2_control_node-3] [WARN] [1753253495.075093206] [FrankaHardwareInterface]: libfranka: unable to set realtime scheduling: Operation not permitted
[ros2_control_node-3] [WARN] [1753253495.075173315] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253496.075278558] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253497.075425445] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253498.075614608] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253499.075778194] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253500.075942498] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253501.076109901] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253502.076270586] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253503.076406495] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253504.076555053] [joint_impedance_controller]: Waiting for valid joint states...
[spawner-6] [WARN] [1753253505.040808856] [spawner_joint_impedance_controller]: Failed getting a result from calling /controller_manager/switch_controller in 10.0. (Attempt 1 of 3.)
[spawner-5] [WARN] [1753253505.067270229] [spawner_franka_robot_state_broadcaster]: Failed getting a result from calling /controller_manager/switch_controller in 10.0. (Attempt 1 of 3.)
[spawner-4] [WARN] [1753253505.073881136] [spawner_joint_state_broadcaster]: Failed getting a result from calling /controller_manager/switch_controller in 10.0. (Attempt 1 of 3.)
[ros2_control_node-3] [WARN] [1753253505.076730755] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253506.076915182] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253507.077034811] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253508.077135412] [joint_impedance_controller]: Waiting for valid joint states...
[... 73 lines cut ...]
[ros2_control_node-3] [WARN] [1753253543.082348903] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253544.082486459] [joint_impedance_controller]: Waiting for valid joint states...
[... cut ...]
[ros2_control_node-3] [WARN] [1753253563.085532682] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253564.085674769] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253565.085884723] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253566.086047225] [joint_impedance_controller]: Waiting for valid joint states...
[ros2_control_node-3] [WARN] [1753253567.086208515] [joint_impedance_controller]: Waiting for valid joint states...'

AndreasJosefGassner, 2025-07-23:
Hi @Jinming-Li, thanks for reaching out!

Just to confirm—did you follow the steps in the README closely, including starting the GELLO publisher first? From your description, it seems like the controller startup might be failing because the `gello-joint-states` topic isn't being received.

If the issue persists, could you please share a few more details about your setup? For example, are you using the dev container, or running everything directly on your host system? That would help us better understand and reproduce the problem.

Looking forward to your update!

RunyangLi, 2025-07-24:
While I run publisher:

(gello) admin123@admin123-System-Product-Name:~/gello_software/ros2$ ros2 launch franka_gello_state_publisher main.launch.py config_file:=example_fr3_config.yaml
[INFO] [launch]: All log files can be found below /home/admin123/.ros/log/2025-07-24-21-20-22-157426-admin123-System-Product-Name-31432
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gello_publisher-1]: process started with pid [31433]
[gello_publisher-1] [INFO] [1753363222.332524596] [gello.gello_publisher]: Auto-detected com_ports: ['/dev/serial/by-id/usb-FTDI_USB__-__Serial_Converter_FTAA0A1T-if00-port0']

It just hold on this place, what is the problem?
Many thanks for your reply!

AndreasJosefGassner, 2025-07-25:
> While I run publisher:
>
> (gello) admin123@admin123-System-Product-Name:~/gello_software/ros2$ ros2 launch franka_gello_state_publisher main.launch.py config_file:=example_fr3_config.yaml [INFO] [...]

I'll respond directly in [the issue thread](https://github.com/wuphilipp/gello_software/issues/50#issue-3260150088).

Aside from that, it appears the problem has been resolved, so I'll go ahead and close this issue.

closed 2025-07-25 by AndreasJosefGassner