wuphilipp/gello_software #50: The ros2 publish problem

RunyangLi, 2025-07-24:
While I run publisher:

(gello) admin123@admin123-System-Product-Name:~/gello_software/ros2$ ros2 launch franka_gello_state_publisher main.launch.py config_file:=example_fr3_config.yaml
[INFO] [launch]: All log files can be found below /home/admin123/.ros/log/2025-07-24-21-20-22-157426-admin123-System-Product-Name-31432
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gello_publisher-1]: process started with pid [31433]
[gello_publisher-1] [INFO] [1753363222.332524596] [gello.gello_publisher]: Auto-detected com_ports: ['/dev/serial/by-id/usb-FTDI_USB__-__Serial_Converter_FTAA0A1T-if00-port0']

It just hold on this place, what is the problem?
Many thanks for your reply!

AndreasJosefGassner, 2025-07-25:
Hi @RunyangLi,

Have you checked whether the `/gello/joint_states` topic is publishing any messages? The logs look good—there doesn’t appear to be an issue. The node starts up and begins publishing as expected.

You can verify this by running:

```bash
ros2 topic echo /gello/joint_states
```

This will allow you to check if joint states are being received.

After confirming that, please follow the steps outlined in the README to continue. Specifically, you’ll need to:

- Launch the **joint impedance controller** to control the robot, and  
- Start the **gripper manager** if you want to control a gripper as well, using the input received from the Gello state publisher.

AndreasJosefGassner, 2025-08-05:
Hi @RunyangLi, just following up — were you able to test it again? If everything looks good, we can go ahead and close this. If there's no update in the next few days, I'll plan to close the issue for now, but feel free to reopen it anytime if needed. Thanks!